hotosm/drone-tm #432: Having a list of compatible, non-compatible and unkown compatibility for drone brands/models

emi420, 2025-01-08:
It would be nice to have a list of drone brands and models, identifying those which:

1. Are compatible with Drone TM (because they have waypoints support and it's possible to copy the KMZ to the device for replacing it)
2. Are not compatible (don't have waypoints support, it's not possible to replace the KMZ or another reason)
3. We don't know if they're compatible, but we would like to know (because are cheap and available)

spwoodcock, 2025-01-08:
Looks like this was done recently in https://github.com/hotosm/drone-tm/pull/402 !

Do you think this info makes sense in the FAQ?

Perhaps it would be more visible if we referenced the FAQ info in the README?

emi420, 2025-01-08:
I think this is very important information! IMO it should be more visible for users, maybe the FAQ page could be accessible directly trough a "FAQ" link in dronetm.org

spwoodcock, 2025-01-08:
There is a link to the docs at the top right:

[image]

I just made a PR to set the docs index page to match the README, and then added to the README:

[image]

Hopefully this is enough to make it visible!

Otherwise we could also update the frontend to have a link to the specific FAQ question if needed

spwoodcock, 2025-06-06:
Perhaps a sidebar link? Any ideas?

Feel free to PR too if you think of something 👍

PR #433 (2025-01-08): Add link to supported drones in README + expand FAQ details
## What type of PR is this? (check all applicable)

- [ ] 🍕 Feature
- [ ] 🐛 Bug Fix
- [x] 📝 Documentation
- [ ] 🧑‍💻 Refactor
- [ ] ✅ Test
- [ ] 🤖 Build or CI
- [ ] ❓ Other (please specify)

## Related Issue

Fixes #432

## Describe this PR

- Made the README page the index for the docs.
- Added a link to the FAQ with drone info.
- Updated FAQ info about supported drones.

## Checklist before requesting a review

- 📖 Read the HOT Code of Conduct: <https://docs.hotosm.org/code-of-conduct>
- 👷‍♀️ Create small PRs. In most cases, this wil [... cut]