hotosm/drone-tm #436: Add align point cloud for aligning multiple flights

smathermather, 2025-01-09:
With https://github.com/hotosm/drone-tm/pull/399, we get group processing, but no approach yet to align subsets of data to each other within a project.

Suggest adding adjacent point clouds to ensure alignment of each group, similar to this write up:
https://docs.opendronemap.org/map-accuracy/#improving-relative-accuracy

While alignment can also be done on DSM or DTM, LAZ file is more precise.

Feature would build on https://github.com/hotosm/drone-tm/pull/343

ivangayton, 2025-01-09:
Agree totally, and LAZ is also preferable because we can use the 3D products

nrjadkry, 2025-01-10:
Hi @smathermather  and @ivangayton ,

Could you please help me understand what the alignment file (LAZ file) is? What is its purpose, and why is it needed?

smathermather, 2025-01-10:
GNSS errors for commodity sensors are rarely sub-meter, but with high resolution datasets, we can easily see offsets in the decimeter range. So for datasets flow during a change in atmospheric conditions or across multiple days, or just unlucky ones, we will see inter-dataset offsets.

Using the point cloud from one flight allows us to align a batch of data with an adjacent flight to ensure consistency in spite of accuracy.

https://docs.opendronemap.org/arguments/align/