hotosm/drone-tm #509: Side overlap recommendations

smathermather, 2025-03-31:
[image]

Side overlap recommendations seem low. Do these recommendations vary based on products needed? They might be fine for orthophoto, but would be low for terrain models.

spwoodcock, 2025-04-15:
Very good point 👍 

We could update the recommendations based on what final output the user selects.
We could also provide pointers to help users assess.

- Orthophoto only: recommend 75% front 65% side as above.
- Terrain models / 3D: recommend 85% front, 85% side?

Do you have any other suggestions for good defaults? 🙏 

We should probably provide a callout box or link to provide further guidance for users:

- Higher overlap improves vertical accuracy. This is crucial for elevation and 3D models.
- Images including lots of low-texture terrain (e.g. sand, water, vegetation) requires higher overlaps (to ensure sufficient keypoint matches, i.e. the points that can be easily matched between images like corners, edges, or textured spots).
- Hilly or mountainous terrain benefits from increased overlaps and/or oblique imagery.
- Side overlap below 70% is not recommended for terrain model generation, as it limits the number of viewing angles for triangulation.

smathermather, 2025-06-20:
It'll depend a bit on flight height as well. You can get away with 75/65 for orthophotos as long as the stuff you're flying over is relatively short and you are at your maximum legal height of 120m. As soon as you fly lower, your minimum overlap might need to be 72-75%.

I never recommend dropping below 70% as a default. It will work 90% of the time, and then other 10% will be very frustrating. I usually use 72% as my floor and then adjust upward based on use case.

spwoodcock, 2026-09-20:
To clarify, I think the points made above were specifically referencing front overlap (as the most important param of the two?).

Also, note from discussion with @ivangayton "At some point I want to design adaptive overlap into the DroneTM. Something like 85% over high urban, 75% over most stuff, and 65% over flat periurban or rural." (also referencing front overlap).

Currently we use a 75% recommended front overlap in DroneTM (and 65% side overlap).