hotosm/drone-tm #513: Automatically detect if there are gaps in the first pass ortho and allow the user to create a second-pass flight plan

spwoodcock, 2025-04-03:
### Issue

- Same as https://github.com/hotosm/drone-tm/issues/512 for if we have missed 'gaps' in the imagery for various reasons.
- Ideally we automatically detect this and flag the areas to the user.

### Solution

- We need a nice solution to automatically detect these missing zones.
- We can calculate the points where the photos **should** have been taken in each flight plan, then get the GPS coord of the actual images we have and compare the two. If there is a missing image based on a threshold / radius around the expected points, then we flag as a missing image.
- Pool all these missing points together, then generate an optimised flight plan to cover the missed parts.

### Considerations

- The user generated a flight plan with certain params: rotation angle, gimbal angle, etc.
  - Currently we don't save these.
  - But if we need to rerun a certain part of the flight plan, we may need to save these params so we can generate a duplicate flight plan covering the missed area (same angle, rotation, etc).

allisonsibrian, 2026-01-22:
Hi,

I am interested in tackling this issue next if it's open for contributions! Working on the tail removal calculation logic was fun, and I want to continue expanding on that foundation! 😸

I've been thinking about the logic for reconstructing the flight plan. Having the new classification workflow is a huge help here, as we can utilize the filtered imagery to: 

- Calculate the azimuth and distance moved for the assigned images after classification, then use a global median as a baseline for what the flight should have looked like, assuming the standard lawnmower pattern.
- Identify any 'long' edges between images where the distance significantly exceeds the baseline to find missing photos in a row.
- Utilize the task AOI to compare our actual coordinates against the expected grid to find the missing imagery.
- Any identified gaps could be 'stamped' as suggested coordinates for a second-pass flight plan.

Would it also be best to add this to image_logic.py, or is it better to create a new module for this?

Let me know if this proposed logic is relevant and if there are any specific references or logic helpful to note here!

spwoodcock, 2026-01-22:
Sounds like a wonderful plan - well thought out 🙌

Once we have an endpoint to return coordinates for a new flight, we could use a threshold to determine if the new flightplan should be waypoints (<100 approx points) or waylines (if more than 100).

All of this will eventually be wrapped in a UI 'Identify Gaps', where either the user can flag specific areas to refly, or this automated logic will identify the areas. The endpoint should probably return the new waypoint flight geojson, so we can display it on a map 😃

Having this piece is really the cherry on the cake! It would solve so many of the issues we had for the Mt Agung Escape Route analysis months ago