hotosm/drone-tm #524: Straighten waylines

ivangayton, 2025-04-07:
# Waylines can be made to not wiggle without extra waypoints

We have been doing a lot of extra work, and adding lots of extra waypoints, to avoid the DJI Mini 4 Pro wanting to fly in a curved path. @smathermather and I have been experimenting with the WayPoint Markup Language (.wpml) to avoid this, and we've found the following parameters that seem to produce nice rectilinear flight plans:

```
<?xml version='1.0' encoding='UTF-8'?>
<kml
	xmlns="http://www.opengis.net/kml/2.2"
	xmlns:wpml="http://www.dji.com/wpmz/1.0.2">
	<Document>
		<wpml:missionConfig>
			<wpml:flyToWaylineMode>safely</wpml:flyToWaylineMode>
			<wpml:finishAction>goHome</wpml:finishAction>
			<wpml:exitOnRCLost>goContinue</wpml:exitOnRCLost>
			<wpml:executeRCLostAction>hover</wpml:executeRCLostAction>
			<wpml:globalTransitionalSpeed>2.5</wpml:globalTransitionalSpeed>
			<wpml:globalRTHHeight>118.8</wpml:globalRTHHeight>
			<wpml:droneInfo>
				<wpml:droneEnumValue>68</wpml:droneEnumValue>
				<wpml:droneSubEnumValue>0</wpml:droneSubEnumValue>
			</wpml:droneInfo>
		</wpml:missionConfig>
		<Folder>
			<wpml:templateId>0</wpml:templateId>
			<wpml:executeHeightMode>relativeToStartPoint</wpml:executeHeightMode>
			<wpml:waylineId>0</wpml:waylineId>
			<wpml:distance>0</wpml:distance>
			<wpml:duration>0</wpml:duration>
			<wpml:globalWaypointTurnMode>toPointAndStopWithDiscontinuityCurvature</wpml:globalWaypointTurnMode>
			<wpml:globalUseStraightLine>1</wpml:globalUseStraightLine>
			<wpml:autoFlightSpeed>2.5</wpml:autoFlightSpeed>
			<Placemark>
```

Specifically note these lines:

```
<wpml:globalWaypointTurnMode>toPointAndStopWithDiscontinuityCurvature</wpml:globalWaypointTurnMode>
<wpml:globalUseStraightLine>1</wpml:globalUseStraightLine>
```

In our .wpml previously, the `globalUseStraightLine` parameter has been set to 0. I've set it to 1, and I get a square-looking flight path.

I haven't actually flown these yet as it's raining, but in the controller it shows a square path, whereas before it showed a wiggly path.
 
Pending some actual tests, I suggest we modify our mission config parameters to this setting. 

With luck, we might be able to stop using the extra waypoints to avoid the rotated photos!

spwoodcock, 2025-07-04:
Extra context from Slack, explaining why we have flight plans with two points at the end of each wayline:

[image]

"""
With the DJI drones, without those second points we get those figure 8 flight paths that we first observed in Bali. I'm pretty sure there's a different setting that we can modify in the flight plan header to mitigate this, but for the moment the extra points do the trick very robustly.

There are a ton of things that we're doing just to compensate for the frustrating limitations of the DJI drones... controller app can't display more than 100 waypoints—even though the drone itself is perfectly capable of handling them.

The two points at each end of each wayline are another thing we have to do because we're fundamentally trying to get a system that wasn't built to do mapping. A few months ago, I did find a setting that could be modified in the wpml that made those extra waypoints unnecessary in a couple of toy test flights that I did, but we'd have to test out a little more extensively to be sure that it works robustly on all flight plans before we make that change globally.
"""
The extra settings are described in the description above! ^^^

spwoodcock, 2025-07-04:
Watched some videos and did some extra research here: those two params combined seem to be the settings we are looking for 👍 

I think there is no harm enabling `globalUseStraightLine=1` by default, but then at a later point we can test if removing the extra waypoint at the end of waylines is possible, while still retaining straight lines and good imagery.

commit eff47f75 (2025-07-04): fix(drone-flightplan): enable globalUseStraightLine=1 by default, fixes #524
closed 2025-07-04 by spwoodcock