hotosm/drone-tm #525: How to import DJI Skyroute Library?

Bin-ze, 2025-04-07:
I tried to generate kmz file using this repository, but templete.kml file was missing, so it didn't meet DJI's route planning format

So I started trying to manually add templeta.kml file and merge the generated waylines.wpml into .kmz, but it still failed when imported into DJI Sky2.

This is good work, but it is almost difficult to use for someone who doesn't know much about this field. Lack of documentation, lack of examples, lack of tutorials, can you help me?

spwoodcock, 2025-04-15:
Hi @Bin-ze! Thanks for reaching out 😄 

Could you clarify what you mean by 'DJI Skyroute' and 'DJI Sky2'?
I can't find any references to these online.

Do you mean the DJI Air 2S drone?

Bin-ze, 2025-04-16:
DJI Sky2： https://enterprise.dji.com/cn/flighthub-2， But I have already solved these problems. The problem now is that I think some calculations in the code are wrong. See： https://github.com/hotosm/drone-flightplan/pull/28

spwoodcock, 2025-04-16:
Ah, thanks for flagging! I will move that PR progress across to the DroneTM repo!!

closed 2025-04-16 by spwoodcock