hotosm/drone-tm #526: calculate_parameters func: Fixed Parameters

Bin-ze, 2025-04-07:
calculate_parameters func have some fixed parameters: 
    Vertical FOV = 0.71
    Horizontal FOV = 1.26
I compute this using:
```
  Horizontal FOV = Image_W / Image_focal

  Vertical FOV = Image_H / Image_focal

  GSD_to_AGL_CONST = Focal * Image_W / Sensor_w * 100, Folcal and Sensor_w using mm for units.
```

I want to know if my calculation formula is correct. 
If it is correct, I cannot get the fixed value you calculated from this calculation method. 
If it is wrong, what should be the correct calculation method?

Bin-ze, 2025-07-07:
I think you are still wrong, because the equivalent focal length is relative to a full-frame camera: 36×24mm.

smathermather, 2025-07-07:
The following is a useful check as I've used this calculator in the past extensively with great success:
https://www.scantips.com/lights/fieldofview.html

Also good to cross check with a flight planner and manually enter the camera specs, e.g. [Mission Planner](https://ardupilot.org/planner/docs/mission-planner-flight-plan.html).

This project is also a fairly reliable footprint calculator:
https://github.com/spifftek70/Drone-Footprints

And you can also double check against individually orthorectified images if you want to get really fancy:
https://github.com/OpenDroneMap/ODM/tree/master/contrib/orthorectify

(The first link is probably enough)

Bin-ze, 2025-07-08:
```
sensor_width = 36.0  # mm，全画幅宽度
focal_length = 24.0  # mm，等效焦距

fov_horizontal = 2 * np.arctan(sensor_width / (2 * focal_length)) * 180 / np.pi
print(f"水平视场角: {fov_horizontal:.2f}°")
```
I think this is correct

Bin-ze, 2025-07-08:
@spwoodcock

spwoodcock, 2025-07-12:
Thanks both!

I think I finally cracked it (correct me if I'm wrong) 😄 

formulas
```python
def _calculate_constants(
    sensor_height_mm: float,
    sensor_width_mm: float,
    equiv_focal_length_mm: float,
    image_width_px: int,
):
    """A helper function to calculate constants for a new drone added."""
    # Sensor calcs
    sensor_diagonal_mm = math.sqrt(sensor_width_mm**2 + sensor_height_mm**2)
    sensor_crop_factor = 43.27 / sensor_diagonal_mm
    actual_focal_length_mm = equiv_focal_length_mm / sensor_crop_factor

    # For degrees, simply: horizontal_fov_rad * 180 / math.pi
    horizontal_fov_rads = 2 * math.atan(sensor_width_mm / (2 * actual_focal_length_mm)) 
    vertical_fov_rads = 2 * math.atan(sensor_height_mm / (2 * actual_focal_length_mm))
    # mm --> cm to match GSD in cm/px (the *100 in formula)
    gsd_to_agl_const = (actual_focal_length_mm * image_width_px) / (sensor_width_mm * 100)

    print(f"Actual focal length:  {actual_focal_length_mm:.2f}mm")
    print(f"Horizontal FOV:       {horizontal_fov_rads:.2f}rad")
    print(f"Vertical FOV:         {vertical_fov_rads:.2f}rad")
    print(f"GSD To AGL Const:     {gsd_to_agl_const}")
```

specs
```python
DRONE_STATS = {
    DroneType.DJI_MINI_4_PRO: {
        # 1/1.3-inch CMOS
        # 4:3 (or 16:9 cropped)
        "sensor_height_mm": 7.2,
        "sensor_width_mm": 9.6,
        "equiv_focal_length_mm": 24,
        "image_width_px": 4032,
    },
    DroneType.DJI_AIR_3: {
        # 1/1.3-inch CMOS
        # 4:3 (or 16:9 cropped)
        "sensor_height_mm": 7.2,
        "sensor_width_mm": 9.6,
        "equiv_focal_length_mm": 24,
        "image_width_px": 4032,
    },
    DroneType.POTENSIC_ATOM_2: {
        # 1/2-inch CMOS
        # 4:3 (or 16:9 cropped)
        "sensor_height_mm": 4.80,
        "sensor_width_mm": 6.40,
        "equiv_focal_length_mm": 26,
        "image_width_px": 4608,
    }, 
}
```

final calcs
```python
DRONE_PARAMS = {
    DroneType.DJI_MINI_4_PRO: {
        "VERTICAL_FOV": 0.99,
        "HORIZONTAL_FOV": 1.25,
        "GSD_TO_AGL_CONST": 27.95,
        "OUTPUT_FORMAT": "DJI_WMPL",
    },
    # NOTE Mini 4 Pro, Mini 3 Pro, and Air 3 all have same CMOS size
    # and focal length, so have the same stats here
    DroneType.DJI_AIR_3: {
        "VERTICAL_FOV": 0.99,
        "HORIZONTAL_FOV": 1.25,
        "GSD_TO_AGL_CONST": 27.95,
        "OUTPUT_FORMAT": "DJI_WMPL",
    },
    DroneType.POTENSIC_ATOM_2: {
        "VERTICAL_FOV": 0.93,
        "HORIZONTAL_FOV": 1.17,
        "GSD_TO_AGL_CONST": 34.61,
        "OUTPUT_FORMAT": "POTENSIC_SQLITE",
    },
}
```

closed 2025-07-12 by spwoodcock