hotosm/drone-tm #534: Add setup / easy workflow for processing imagery on a local laptop, prior to ortho upload

spwoodcock, 2025-04-15:
## Issue
 
- In some scenarios the user doesn't have great upload bandwidth for the raw imagery. 
- We can scale the backend NodeODM processors (machines / nodes) all we want, but if the user can't upload the imagery easily then this is a key failure point.

## Solution

- A simple one command solution that sets up a **WebODM** instance on the users device, then opens it in their web browser.
  - Admin user should be configured / auth disabled if possible.
- Provide some simple instructions for how to use WebODM to process imagery.
- The user can then process everything on their local machine and upload the final ortho products only.

## Considerations

- Obviously this isn't an ideal setup, as we may lose the raw imagery in the long run (stuck on the users machine), but if it's the only way, it's the only way!
- At least we get a useful final ortho that can be accessed by others.

spwoodcock, 2025-10-07:
We have been discussing in the field, and considering our use of QField, perhaps we could integrate this into a QField plugin for photo validation / pre-processing?

SColchester, 2026-09-17:
From Ghana experience, best if this is very lightweight - just needs the task boundaries and the location points of the jpegs, is there any way of dragging jpegs into an interface on a laptop / webpage and it presents only location points, and only extracts / uploads the location points rather than all the image data?