hotosm/drone-tm #599: Remove flightplan tails during post processing

spwoodcock, 2025-09-22:
## Problem Summary

- We encountered some issues during processing of Freetown imagery.
- The first 20-30 images are taken from the take off point to the first waypoint position.
- This is because we need to use the interval photo setting from the take off point (detailed waypoint missions with photo capture at each point would not have this issue, but in order to fit the whole mapping area in the DJI Fly app, we simplify the waypoints to 'waylines' - essentially waypoints for the start and end of each line to fly, plus a photo interval).
- These tails should be removed in an automated way, prior to final imagery processing (or perhaps even prior to imagery upload if at all possible, but less likely).

## Documenting points made by @smathermather:

- "Tilt in the models themselves... The following is a difference map with a reference low resolution terrain model, and we can see some different variance from the low resolution model in the corners of 058 and 056 and probably others":

[image]

- "When we've seen this previously, we've selectively removed the images that are collected en route and the tilt goes away, but this is a fairly large subset of the total data (~30k images), so probably good to take a first stab at automated removal."

- "If we look at our flights, we can see these little tails from the takeoff point to the main part of the flight plan":

[image]

[image]

- "So, we need to identify and remove these and hope this fixes our issue"

## Possible solution from @smathermather 

- "So, we pick each 2 photos and calculate a line between them so we can get the azimuth of that line":

[image]

- "But, we probably only need to filter on the first 30 images, which will make any angle/azimuth based filtering safer, so we subset and look at that angle":

[image]

And remove anything that isn't near 90-degrees:

[image]

It's not perfect, but probably good enough:

[image]

- SQL algorithm that could achieve this:

```sql
DROP TABLE IF EXISTS freetown.freetown_flight_lines_start;
CREATE TABLE freetown.freetown_flight_lines_start AS
	WITH beginning AS (
		SELECT
		  * 
		FROM (
		  SELECT
			ROW_NUMBER() OVER (PARTITION BY flight ORDER BY datetime) AS row,
			t.*
		  FROM
			freetown.freetown_flights t) x
		WHERE
		  x.row <= 30
		),
	ordered AS (
		SELECT
			id,
			flight,
			datetime,
			geom,
			LAG(geom) OVER (PARTITION BY flight ORDER BY datetime) AS previous_geom,
			LAG(datetime) OVER (PARTITION BY flight ORDER BY datetime) AS previous_datetime
		FROM
			beginning
		),
	lines AS (
		SELECT id, flight, previous_datetime, datetime, degrees(ST_Azimuth(previous_geom, geom)) AS azimuth, 
			ST_MakeLine(previous_geom, geom) AS geom FROM ordered
		)
	SELECT * FROM lines WHERE geom IS NOT NULL;
```

allisonsibrian, 2026-01-07:
Hi, I am interested in picking up this issue if its still available to tackle!

I have been looking at the proposed comments and have a few clarifying questions:
1.  Instead of using a fixed 90-degree filter, would it be better to look for the first significant change in the azimuth between images? This would help identify when the drone turns into the first waypoint regardless of which direction the drone is flying (like West or diagonally).
2. For the Freetown imagery, the first 30 images were filtered. What would be a good number of images to analyze generally?
3. Does it make sense to trigger the removal of the flight tails within the IMAGE_PROCESSING_START event?
4. Is there anything else (code or logic) I should look at as a reference before starting? 

Thanks!

allisonsibrian, 2026-01-08:
Got it, thanks so much for the breakdown! 
Hoping to start tackling this issue this weekend!

closed 2026-01-18 by spwoodcock